dwyl/iot-decibel-meter #11: Using sparkfun sound detector

naazy, 2017-10-10:
http://toribeiro.com/articles/2016-01/sound-level-meter-using-esp8266-arduino-sound-detector

naazy, 2017-11-17:
Converting voltage -> decibels: https://forum.arduino.cc/index.php?topic=376308.0

pitter7488258450-cmyk, 2026-01-11:
Hey everyone! If anyone is still troubleshooting the calibration or just needs a quick way to verify the readings they're getting from the SparkFun sensor against a 'real-world' reference, I found this web tool- [Noise Meter](https://noisemeter.co/) It’s been pretty handy for me to quickly check ambient noise levels using my phone/laptop mic without having to pull out a dedicated handheld meter every time. Might help some of you dial in those threshold values!