dwyl/iot-decibel-meter #15: Adjusting sensitivity of Sparkfun sound detector

naazy, 2017-12-15:
The sparkfun sound sensor was detecting noise right next to do it but not the overall volume of noise in the room. 

As per the [hook up guide](https://electronics.stackexchange.com/questions/269005/how-to-hook-raspberry-pi-with-electret-microphone-to-detect-sound-levels-in-db), I tried to increase the sensitivity of the sound sensor by removing R3 and filling R17 with a through hole resistor but I accidentally pressed down too hard and ended up melting the sound sensor :cry:!
[image]

iteles, 2017-12-15:
🙈 😊  *sad trombone* 🎺

naazy, 2017-12-22:
After writing the code in the arduino IDE #16 and plugging the arduino into the wall, we did not need to adjust the sensitivity of the sound detector.

pitter7488258450-cmyk, 2025-12-31:
Nice save on the software side! Hardware sensitivity mods can be such a pain, especially on those SparkFun boards. It's good to know the Arduino code was enough to get the range you needed for the room volume.